ArduPilot/MissionPlanner #766: MP - increasing lag over time - distancebar related?

GrahamDyer, 2015-01-03:
We're flying long missions, sometimes over two hours and we're seeing with v1.3.17 that the Flight Data screen becomes slower and slower over time with noticeable lag - all controls, the HUD and all reactions are slow while in the FD screen. This is quite serious as MP progressively becomes usable as time passes.

After 45 minutes everything is starting to be noticeably slow and by 1hr30m the HUD no longer updates and every response is extremely laggy. On the Flight Planner screen the lag is gone but returns immediately when going back to the FD screen. Directly after MP restarts everything is fine but one can't restart it during a mission.

Is this related to the new distance bar? It seems so but we can't be sure.

meee1, 2015-01-04:
graham, does playing back the long log cause the same issue? just want to know is its something to do with being connected to real hardware or is just playing back a log does it as well.

GrahamDyer, 2015-01-04:
Will check on another PC and get back to you as I've reverted back to .16

GrahamDyer, 2015-01-04:
A tlog without waypoints (not read before takeoff) runs fine, a tlog with embedded waypoints (read before takeoff) slows to a crawl after 45 mins. PM me an email address if you would like this tlog?

meee1, 2015-01-05:
should be fixed in the latest beta, here 490f1e9c

closed 2015-01-05 by meee1

GrahamDyer, 2015-01-06:
Fixed, works fine now, thank you so much for the prompt attention.