ArduPilot/MissionPlanner #767: DistanceBar resets to 0 when "Set WP"

SSUAV, 2015-01-04:
During a mission when a manual change of next WayPoint via the Set WP on the Actions Tab the DistanceBar resets to 0

meee1, 2015-01-05:
it should fix itself in 5 seconds, as that is the update interval on that bar.

SSUAV, 2015-01-06:
Yes, you are correct. On closer inspection our mission had a couple of DO_JUMP commands back to WP1 which set the number back to zero.
So the white distance number is not the true distance the aircraft has travelled. Maybe display "Dist Travelled (m)" variable here instead or no number at all.
The white bar indicating where in the mission you are is way cool. 
How about just the white bar passing through yellow dots?

meee1, 2015-01-19:
I might try the yellow dot idea,