ArduPilot/MissionPlanner #771: Joystick Axis and Buttons not loaded (files saved to bad location)

CP44, 2015-01-05:
I noticed after a recent update (Window title now shows 1.3.16 build 1.1.5456.14788) that Joystick settings did not load or save properly.

MissionPlanner.log includes this line:
INFO MissionPlanner.Joystick.Joystick - Saving joystick config files [drive]:\Program Files\MissionPlannerjoystickbuttonsArduPlane.xml S:\Program Files\MissionPlannerjoystickaxisArduPlane.xml (c:\Users\hog\Documents\Visual Studio 2010\Projects\ArdupilotMega\ArdupilotMega\Joystick\Joystick.cs:195) [Base Thread]

Checking Program Files did reveal these 2 files:
MissionPlannerjoystickaxisArduPlane.xml
MissionPlannerjoystickbuttonsArduPlane.xml
that were written with data outside the program's folder.
(MissionPlannerjoystickbuttonsArduPlane.xml, 1155 lines)
(MissionPlannerjoystickaxisArduPlane.xml, 57 lines)

Process Monitor showed a QueryOpen for 
"[drive]:\Program Files\MissionPlanner\joystickbuttonsArduPlane.xml" showing that the Read operations are still being attempted on the old (correct?) path.

CP44, 2015-01-06:
[drive] is used above as the Program Files folder is not on the normal (C:) drive on this computer.  To verify the non-standard drive is not the problem, copied MissionPlanner folder to C:\MissionPlanner and received the following error while attempting to Save the joystick settings.

This error was also presented in a pop-up that asked if I would like to report the error.  Sending the report failed, and the program said this is generally due to a lack of Internet connectivity although I have been online the entire time and have not dropped connectivity since creating this Issue report.

2015-01-05 19:33:38,570  INFO MissionPlanner.Joystick.Joystick - Saving joystick config files C:\MissionPlannerjoystickbuttonsArduPlane.xml C:\MissionPlannerjoystickaxisArduPlane.xml (c:\Users\hog\Documents\Visual Studio 2010\Projects\ArdupilotMega\ArdupilotMega\Joystick\Joystick.cs:195) [Base Thread]
2015-01-05 19:33:39,234 DEBUG MissionPlanner.Utilities.Tracking - v=1&tid=UA-43098846-1&cid=abccabd1-0486-4584-8f6b-acc4a161b1a7&t=exception&an=Mission+Planner&av=1.3.16&cd=MissionPlanner.Joystick.JoystickSetup&dp=MissionPlanner.Joystick.JoystickSetup&exd=Access+to+the+path+%27C%3a%5cMissionPlannerjoystickbuttonsArduPlane.xml%27+is+denied.+++at+MissionPlanner.Joystick.Joystick.saveconfig()+in+c%3a%5cUsers%5chog%5cDocuments%5cVisual+Studio+2010%5cProjects%5cArdupilotMega%5cArdupilotMega%5cJoystick%5cJoystick.cs%3aline+200%0d&exf=0&cd5=System.UnauthorizedAccessException%3a+Access+to+the+path+%27C%3a%5cMissionPlannerjoystickbuttonsArduPlane.xml%27+is+denied.%0d%0a+++at+System.IO.__Error.W&ul=en-US&sd=32-bits&sr=1920x1080&cd1=Microsoft+Windows+NT+6.1.7601+Service+Pack+1&cd4=8&z=1845240270 (c:\Users\hog\Documents\Visual Studio 2010\Projects\ArdupilotMega\ArdupilotMega\ExtLibs\Utilities\Tracking.cs:275) [20]

meee1, 2015-01-06:
please update to 1.3.17, this fixed the path issue

closed 2015-01-06 by meee1

CP44, 2015-01-06:
My apologies for missing that fact; *.16 didn't show the prompt for the Christmas update.  After telling it to check for updates in Help tab, the updater got *.17 without issue.  Many thanks, as always!