ArduPilot/MissionPlanner #774: Survey Grid V1 distance calculations and time calculations incorrect

jmachuca77, 2015-01-06:
In the latest Beta the calculations for distance and time are way off. In this example the flight should take minutes, and the perimeter of the area is 1km long so there is no way the distance flown is 100km.

[image]

jmachuca77, 2015-01-07:
Has this been pushed out to the latest beta? I tried and it now works fine until you change the flying speed, then it calculates wrong again. But I am not sure if the beta (1.1.5484.14078) I have has the change yet.

meee1, 2015-01-07:
the problem is actualy around the wait at wp settings. try changing that to 0, and see if it makes sence