ArduPilot/MissionPlanner #775: Rally Points Visibility

Craig3DRobotics, 2015-01-07:
Rally points allow the user to set safe landing points for the vehicle in cases where the home location is not a safe area to land.

According to Tridge, here's how rally points work in the auto pilot:

1) RTL will normally go to the nearest rally point if there is a rally 
point set. If there is no rally point set it will go home.

2) if the nearest rally point is more than RALLY_LIMIT_KM away and if 
home is closer than the nearest rally point then the rally point 
won't be used, and instead home will be used. On copter 
RALLY_LIMIT_KM defaults to 2 km.

The reason rally points are important for fixed wing is that returning 
to and circling above 'home' in a fixed wing can be both dangerous and 
against club rules. At my club it would get me thrown out, as it would 
at many/most flying clubs in Australia. For safety you have to setup a 
rally point for each of your flying fields in a safe location.

For Droidplanner:

When a vehicle is going towards a rally point instead of home, the user should be alerted.
The rally point should show up on the flight screen and the editor screen
The user should be able to reset / set rally points for the vehicle in the editor.

Craig3DRobotics, 2015-01-07:
Supplemental: The RALLY_TOTAL should be requested from the vehicle as soon as possible after connecting to the vehicle. If the value is non zero then the rally point locations should be requested from the vehicle and displayed to the operator.
http://copter.ardupilot.com/wiki/arducopter-parameters/#Rally_Total_RALLY_TOTAL

meee1, 2015-01-07:
currently when you download a flight plan from the AP, rally point are downloaded as well

meee1, 2015-01-07:
done 9ca30144

closed 2015-01-07 by meee1