ArduPilot/MissionPlanner #784: Survey tool issues v1.3.18

francisco3dr, 2015-01-28:
Tried to make a survey grid on this section.

[image]

It was MP v1.3.17, it works fine on this version.

If I try to make the same mission on MP 1.3.18 it will yield this result.

[image]

meee1, 2015-01-28:
please try updating to the beta. if all is good I will push a new mp version

francisco3dr, 2015-01-28:
Ok, so it seems that the Beta version will allow me now.

[image]

Thank you for the help.

closed 2015-03-05 by meee1