ArduPilot/MissionPlanner #790: Can't graph multiple series from tlog reliably

ryancu7, 2015-02-02:
The graph tool (Telemetry Logs > Tlog > Kml or Graph > Graph Log)  doesn't seem to work as well as it used to. (I'm now using Mission Planner 1.3.19). 

It can't seem to display the parameters for two different message types at the same time. E.g.
1. Load tlog of a flight into graph window
2. Left click GLOBAL_POSITION > alt checkbox  (graph displays correctly :smiley: )
3. Right click VFR_HUD > alt checkbox (graph displays correctly :smiley: y-axis on right side of graph has the same range as left side)
4. Right click VFR_HUD > alt again to deselect it (the whole graph goes blank :disappointed: It should keep GLOBAL_POSITION > alt)
5. Left click GLOBAL_POSITION > alt to turn it off (nothing happens because it is already brokenly off)
6. Left click GLOBAL_POSITION > alt to turn it on (graph displays correctly :smiley: )
7. Left click VFR_HUD > alt (appears as a flat line at the bottom, even though we saw earlier that it has the same y-axis range :disappointed: )
8. Left click VFR_HUD > alt label, not checkbox  (the graph doesn't change and now two of them are now listed in the legend :disappointed:)

There seems to be a difference between clicking the checkbox to clicking the label and neither way seems to work properly. 

I wanted to display GLOBAL_POSITION > alt and VFR_HUD > alt with the left axis and

SSUAV, 2015-02-02:
> 3. Right click VFR_HUD > alt checkbox (graph displays correctly  y-axis on right side of graph has the same range as left side)
> 4. Right click VFR_HUD > alt again to deselect it (the whole graph goes blank  It should keep GLOBAL_POSITION > alt)

+1

meee1, 2015-02-04:
the latest beta should fix this.

closed 2015-02-04 by meee1

ryancu7, 2015-02-04:
OK, thanks, it is better now. However, there's still a problem if double clicking on a checkbox.