ArduPilot/MissionPlanner #798: Request: UI changes for display of waypoints

nicksargeant, 2015-02-11:
Hello all. 
I hope I am not the only one what gets confused but when planning complicated missions such as those with many waypoints or when using alternating flight lines, interpreting the mission the screen can become very difficult: 
[image]
I am by no means an expert when it comes to UI design but I believe there may be a few minor tweaks that could improve the user experience.  Firstly having the waypoint number offset to the top right of the actual waypoint can be confusing when they are bunched: 
[image]
Removing the right hand offset or the separate waypoint number box entirely would make things more readable:
[image] -> [image] OR [image]
Alternatively, perhaps the number boxes should be hidden unless the mouse is hovered over the individual waypoint?
Additionally, It would be nice of there were direction arrows mid waypoint to show the flight direction:
[image]

If implementing this, the UI may break if using alternating flightlines so perhaps arrows are only needed along the long flightlines: 
 [image]
Thoughts?

iskess, 2015-02-11:
+1
Thanks for bringing up this frustration. Most importantly the WP number offset. I would like the number inside the marker. It can be very small and grow bigger if the pointer hovers over it. This will behavior will also indicate that the WP will be moved if clicked rather than a new WP created. 
I like you direction arrow idea. Only on the long legs. We want the arrows to be subtle and thin to avoid more clutter.

nicksargeant, 2015-02-13:
Agreed iskess, by no means to I want to make the UI more cluttered however I believe small, unobtrusive arrows would be of benefit. Ideally you could be able to turn them on/off in the missionplanner configuration.

meee1, 2015-03-05:
please check the latest beta. nothing set in stone yet, but testing ideas

iskess, 2015-03-05:
I'm away from a computer for a few days, Nick, can you post a screen shot of the beta?
Thanks!

nicksargeant, 2015-03-09:
Thank you very much @meee1! I think this greatly increases the ease of interpretation. @iskess, apologies for the delay. I have attached two screenshots, one of the original complex flightplan and  another that is a more conventionally shaped polygon. 
[image]
[image]
What are your thoughts?

nicksargeant, 2015-03-09:
My only comment could be that perhaps black text (instead of red) would offer more contrast for the waypoint numbers?

iskess, 2015-03-09:
I think it is great! Thank you Michael.

closed 2015-03-22 by meee1