ArduPilot/MissionPlanner #800: Survey Grid Missing Mono

HarrisonHeli, 2015-02-16:
Using Mission Planner 1.3.19 from firmware.diydrones.com on mono 3.28 on Ubuntu 14.04.

The survey grid and grid 2 options are still missing in the menu items.
I have run the same release on a windows VM and the option is present and working correctly.

I note that the problem existed with 1.3.7
survey grid missing mono #674 

Is the grid run as a plugin or built into the exe? 
Happy to do more testing if given some pointers.

HarrisonHeli, 2015-02-16:
Have tried the latest mono 3.12.0. Made no difference.

HarrisonHeli, 2015-02-16:
Back again. Discovered something more. 
On Windows 7 in a VM.

If I extract the zip file to a directory and run the enclosed MissionPlanner.exe, The Grid and Grid2 option is missing from the mission planner list. Just like the linux+mono problem I described above.

If I use the MissionPlanner.msi install and then run the exe, the Grid and Grid V2 options are available and working.

meee1, 2015-02-16:
good find, please do an auto update on your Linux install
and it will hopefully fix itself

meee1, 2015-02-16:
double check that in the mp folder there is a 
plugins directory
and in there is missionplanner.grid.dll

HarrisonHeli, 2015-02-16:
Cool. Thanks for the reply.

Ran the Update - This is the one on the Help page called "Check For Updates"?
It appeared to run correctly and reloaded the application at the end.
Unfortunately, The grid options were not available on the menu

I also tried to update to the beta release 1.3.19.2 but there was no change.

I can confirm that the missionplanner.grid.dll  is present in the plugins directory. Here is a list of the files: 

-rw-r--r-- 1 harrisonheli harrisonheli 233984 Feb 16 20:46 GMap.NET.Core.dll
-rw-r--r-- 1 harrisonheli harrisonheli 154888 Jan  5 23:19 GMap.NET.Core.xml
-rw-r--r-- 1 harrisonheli harrisonheli 150528 Sep 11 23:27 GMap.NET.WindowsForms.dll
-rw-r--r-- 1 harrisonheli harrisonheli  36806 Sep 11 23:27 GMap.NET.WindowsForms.xml
-rw-r--r-- 1 harrisonheli harrisonheli 174592 Feb 16 20:46 MissionPlanner.Grid.dll
-rw-r--r-- 1 harrisonheli harrisonheli  77312 Feb 16 20:46 MissionPlanner.Grid.pdb
-rw-r--r-- 1 harrisonheli harrisonheli 113664 Feb 16 20:46 MissionPlanner.Gridv2.dll
-rw-r--r-- 1 harrisonheli harrisonheli  71168 Feb 16 20:46 MissionPlanner.Gridv2.pdb
-rw-r--r-- 1 harrisonheli harrisonheli  62976 Feb 16 20:46 MissionPlanner.SimpleGrid.dll
-rw-r--r-- 1 harrisonheli harrisonheli  50688 Feb 16 20:46 MissionPlanner.SimpleGrid.pdb
-rw-r--r-- 1 harrisonheli harrisonheli   9728 Feb 16 20:46 MissionPlanner.Stats.dll
-rw-r--r-- 1 harrisonheli harrisonheli  19968 Feb 16 20:46 MissionPlanner.Stats.pdb
-rw-r--r-- 1 harrisonheli harrisonheli  81920 Mar 25  2014 ProjNet.dll
-rw-r--r-- 1 harrisonheli harrisonheli 252823 Mar 25  2014 ProjNet.xml
-rw-r--r-- 1 harrisonheli harrisonheli  26624 Feb 16 20:46 TrackerHome.dll
-rw-r--r-- 1 harrisonheli harrisonheli  40448 Feb 16 20:46 TrackerHome.pdb
drwx------ 2 harrisonheli harrisonheli   4096 Feb 16 20:47 zh-Hans

On the zip file installed version. I removed the plugins directory and MissionPlanner.exe ran with no faults or complaints about the missing directory.
I then went to the msi installed version on the Windows VM and deleted the "plugins" dir.
The grid options were no longer available.

I am guessing the zip file version is not seeing the path to the plugins for some reason.

For the zip file installed version. Looking in the MissionPlanner.log file, all I see mentionted about plugins is:
"2015-02-16 22:00:07,137  INFO MissionPlanner.MainV2 - Load Pluggins (:0) [Base Thread]"
"2015-02-16 22:00:07,139  INFO MissionPlanner.MainV2 - Load Pluggins Done (:0) [Base Thread]"

Fot the msi installed version I get 
"INFO MissionPlanner.MainV2 - Load Pluggins (c:\Users\hog\Documents\Visual Studio 2010\Projects\ArdupilotMega\ArdupilotMega\MainV2.cs:2139) [Base Thread]
2015-02-16 22:06:08,534  INFO MissionPlanner.Plugin.PluginLoader - Plugin Load C:\Program Files (x86)\Mission Planner\Plugins\MissionPlanner.Grid.dll (c:\Users\hog\Documents\Visual Studio 2010\Projects\ArdupilotMega\ArdupilotMega\Plugin\PluginLoader.cs:49) [Base Thread]
2015-02-16 22:06:08,534  INFO MissionPlanner.Plugin.PluginLoader - Plugin Init Grid 0.1 by Michael Oborne (c:\Users\hog\Documents\Visual Studio 2010\Projects\ArdupilotMega\ArdupilotMega\Plugin\PluginLoader.cs:60) [Base Thread]
2015-02-16 22:06:08,544  INFO MissionPlanner.Plugin.PluginLoader - Plugin Load C:\Program Files (x86)\Mission Planner\Plugins\MissionPlanner.Gridv2.dll (c:\Users\hog\Documents\Visual Studio 2010\Projects\ArdupilotMega\ArdupilotMega\Plugin\PluginLoader.cs:49) [Base Thread]
2015-02-16 22:06:08,544  INFO MissionPlanner.Plugin.PluginLoader - Plugin Init Grid 0.1 by Michael Oborne (c:\Users\hog\Documents\Visual Studio 2010\Projects\ArdupilotMega\ArdupilotMega\Plugin\PluginLoader.cs:60) [Base Thread]
2015-02-16 22:06:08,554  INFO MissionPlanner.Plugin.PluginLoader - Plugin Load C:\Program Files (x86)\Mission Planner\Plugins\MissionPlanner.SimpleGrid.dll (c:\Users\hog\Documents\Visual Studio 2010\Projects\ArdupilotMega\ArdupilotMega\Plugin\PluginLoader.cs:49) [Base Thread]
2015-02-16 22:06:08,554  INFO MissionPlanner.Plugin.PluginLoader - Plugin Init SimpleGrid 0.1 by Michael Oborne (c:\Users\hog\Documents\Visual Studio 2010\Projects\ArdupilotMega\ArdupilotMega\Plugin\PluginLoader.cs:60) [Base Thread]
2015-02-16 22:06:08,564  INFO MissionPlanner.Plugin.PluginLoader - Plugin Load C:\Program Files (x86)\Mission Planner\Plugins\MissionPlanner.Stats.dll (c:\Users\hog\Documents\Visual Studio 2010\Projects\ArdupilotMega\ArdupilotMega\Plugin\PluginLoader.cs:49) [Base Thread]
2015-02-16 22:06:08,694  INFO MissionPlanner.Plugin.PluginLoader - Plugin Init Stats 0.1 by Michael Oborne (c:\Users\hog\Documents\Visual Studio 2010\Projects\ArdupilotMega\ArdupilotMega\Plugin\PluginLoader.cs:60) [Base Thread]
2015-02-16 22:06:08,714  INFO MissionPlanner.Plugin.PluginLoader - Plugin Load C:\Program Files (x86)\Mission Planner\Plugins\TrackerHome.dll (c:\Users\hog\Documents\Visual Studio 2010\Projects\ArdupilotMega\ArdupilotMega\Plugin\PluginLoader.cs:49) [Base Thread]
2015-02-16 22:06:08,714  INFO MissionPlanner.Plugin.PluginLoader - Plugin Init GPS Tracker Home Plugin 1.0 by Will Bryan"

meee1, 2015-02-16:
when you start MP under mono, what directory do you start it from?, as that me be the cause of the issue

meee1, 2015-02-16:
ok, I know the problem
on your mono install change the dir name from "plugins" to "Plugins"

meee1, 2015-02-16:
modified here 16c1f513

meee1, 2015-02-16:
in the next mp version it will be all lower case, as work around though please do above

HarrisonHeli, 2015-02-16:
Yip. That nailed it! 

Thank you!

closed 2015-02-16 by HarrisonHeli