ArduPilot/MissionPlanner #806: Auto remove the cam shots in the "overshoot" path (dynamic CAM_TRIG_DIST)

robustini, 2015-02-26:
During automatic mission with the "survey grid" generator often have to use a overshoot path to allow the plane to execute the turn and get back on course for the new line.
The CAM_TRIG_DIST parameter is set automatically by the Planner considering many factors, unfortunately is maintained until the end of the mission and in this way produce shots with the camera that are not useful for the photogrammetric purpose based on the defined polygon.
It would be useful to do so in order to set CAM_TRIG_DIST to zero, perform the overshoot and set it back to the previous right value as soon as the plane passes the first point useful to photograph.
This would eliminate many shots taken that would not be necessary.

robustini, 2015-02-26:
[image]

robustini, 2015-10-10:
@meee1: hi Michael, no news about this?

robustini, 2015-10-10:
I hope it's clear my concept, disable the cam trigger in the overshoot area (marked in purple rectangles), as not interested and produces many useless photos.

[image]

iskess, 2015-10-10:
Are you proposing that the Survey tool automatically adds two DO_SET_CAM_TRIGG_DIST commands for each turn?

saruban, 2015-10-11:
i want to change the theme(such as colors) of this mission planner software for my learning purpose. How Can I Do This. Please help me.

robustini, 2015-10-11:
@iskess: exactly!

robustini, 2015-10-12:
I am proposing an option (selectable) of "takes pictures only inside the polygon", many who work with these aircraft are waiting for this option.

meee1, 2015-10-12:
I should be able to add this, but the default will be off

robustini, 2015-10-12:
Hi Michael, thanks!
Of course, the default will be off, but I'm sure the people who work with photomapping there will always put this option enabled.
I'm going to make a video of "APM:Plane V3.4.0" during a photogrammetry with auto takeoff and autolanding with my FX79, with triple visual (FPV, Planner and shoot), but before i wait this add-on.
Bests,

Marco

robustini, 2015-10-27:
Feature "breakup starts" added, thanks @meee1!

closed 2015-10-27 by robustini