ArduPilot/MissionPlanner #807: Georefimage not using relalt even when selected

psinha, 2015-03-02:
Images seem to be getting tagged only with the GPS alt (ie, not fused with IMU/baro data) even when RelAlt is selected. The tagged images have an altitude corresponding to the erratic green line in the plot below, rather than the smooth red line representing RelAlt. Seems like it might be an index number issue?

[image]

meee1, 2015-03-05:
where you using a  tlog or a log?

psinha, 2015-03-05:
tlog.

g00bd0g, 2015-03-05:
Also probably not geotagging using imu fused data for lat/long. Be nice to use fused data for all geotagging data. 

Additionally, not sure if it's worthy of an issue report, but mission planner also uses raw GPS data for kmz/kml flight path generation, they would look better and be more accurate if it used IMU fused data.

meee1, 2015-03-05:
my first question for the original poster is, why are you georefrencing using relative alt? ie it does not make sence?

g00bd0g, 2015-03-05:
psinha posted this on my behalf, I'm following up. The answer to your question "why are you georefrencing using relative alt?" is because it is a much cleaner and more accurate representation of the actual flight path altitude. Just look at the shape of the two curves in the photo, the relalt curve much more closely matches the actual flight path I flew. The raw GPS data is much less accurate.

meee1, 2015-03-05:
this issue looks like it may need to be posted on the ardupilot issue list.
as that data in the relat and alt fields from the same message do not stay in sync, like your picture above.
what happens if you graph gps raw int alt as well?

g00bd0g, 2015-03-05:
Here is a graph with gps raw int alt. Sorry it is from a different tlog because I forgot which one used for the 1st graph. But it still shows the issue.

[image]

meee1, 2015-03-06:
something is not adding up here, how can the relative alt be below and then above within a change of 50m. I think there is another issue at work here.

g00bd0g, 2015-03-06:
We are operating near structures that interfere with the gps signal. That is why the imu fused data is much more accurate especially in this case. But im sure this is present to some degree in all the data I have looked at. I think this can be verified by anyone with a simple test flight and comparing the data as I have done. Id be happy to fly any test profile you think might help to diagnose this issue.  Thanks again for looking at this.

meee1, 2015-05-26:
this should be sorted.
now that there is a now POS message in DF logs, this is a better way to do it.

closed 2015-05-26 by meee1