ArduPilot/MissionPlanner #809: Request: Use IMU fused postional data for kml export

g00bd0g, 2015-03-05:
It seems the Google Earth KML export function uses raw GPS data. I have verified this by checking the the alts of the flight path in Google Earth correspond to the the raw gps alt (altglobal_position_int_t) and not the IMU fused data (relative_altglobal_poistion_int_t-R) . I believe the paths would be smoother and more accurate if we used the IMU fused data.

meee1, 2015-03-05:
kml's from tlogs already use this data.

closed 2015-03-05 by meee1

g00bd0g, 2015-03-05:
I can verify this is not true for altitude at least. The KML altitudes match the raw gps data, not the smooth IMU data. I can provide a sample tlog and kml file if you would like?

Thanks!

g00bd0g, 2015-03-05:
Here is some data to support my position:

Graph showing raw and fused alt data
[image]

Screenshot of kml export flight path. Note center of path (where the mouse hand is) corresponds to the middle altitudes of the graph. See how the IMU fused data shows that as a more or less level path, where the raw gps and kml file show it as a sharply descending path. I can confirm, I even have picture and video from the copter itself, that the path was more or less level.
[image]

If it is intended, as you say, to use the IMU data for KML export, than this is certainly broken or a bug, because I am convinced beyond any doubt, we are using raw GPS data.

Please let me know if I have conveyed this clearly, and/or if I can do anything further to assist?

BTW, if you fix this and the similar georef issue (#807)I will gladly buy you a bottle or 6 pack or your favorite brew :)

g00bd0g, 2015-03-05:
One more thought. I think the general "sloppiness" of the flight path in google earth, shows that it does not match the much more smooth and precise flight of the copter itself. You can see all sorts of zig-zags and rapid altitude change that do match the copters actual flight path.

Thanks!