ArduPilot/MissionPlanner #812: Display all channel output in MP?

GrahamDyer, 2015-03-07:
The Pixhawk supports outputs to more than 8 channels but ever since APM2.x days the MP has always only shown ch 1-8 outputs. Please can the outputs to the additional channels be shown in Status?
[image]

nicksargeant, 2015-03-09:
Yes please.

meee1, 2015-03-09:
there is no message to send these extra channels via mavlink, it will require a change on the autopilot side before can do anything on this

GrahamDyer, 2015-03-15:
Tridge says: "When there are more than 8 input channels ArduPilot will send the RC_CHANNELS message with up to 16 channels. That is in all current releases. The problem is that MissionPlanner doesn't display it.
Please open an issue with MissionPlanner."
https://github.com/diydrones/ardupilot/issues/1968#issuecomment-80198537

meee1, 2015-03-15:
graham, where you after the 16 channels in? or the 16 channels out?

rmackay9, 2015-03-16:
This is the message apparently that is sent by arducopter and arduplane
http://mavlink.org/messages/common#RC_CHANNELS

meee1, 2015-03-16:
yes, that message is for rc inputs only, not outputs, ive added all 16 inputs in commit today d7e9ebdb28c17de2f239a6759d2add81215f1141

GrahamDyer, 2015-03-16:
Personally I'm looking for the ch 9-12 outputs, using the extra channels for light switch and zoom on the camera.

GrahamDyer, 2016-02-22:
Request for the outputs to be enabled for this issue to be resolved?

meee1, 2016-02-22:
still requires copter/plane changes.
a new message created for all 18 channels similar to the rc input eg mavlink_msg_rc_channels_send, that adapts to 18 channels when needed

GrahamDyer, 2017-05-27:
Nothing further on this?

closed 2018-06-15 by GrahamDyer