ArduPilot/MissionPlanner #815: Please add support for RealFlight Interlink Elite Controller

R-Lefebvre, 2015-03-16:
Purchased RealFlight 7.5 just to get the Interlink controller to do functional testing of Arducopter.  Currently it works as a Joystick, and the analog inputs for Ch1-4, and the Ch6 knob work as expected.

However, the Ch 5, 7 and 8 switches do not work as analog inputs, they are joystick "buttons".

Through testing, I have determined Ch5 is button input 0, down is off, up is on.  It would be great if we could slave this artificially to the Ch5 input.  When button 0 is off, Ch5 PWM is 1000, when it is on, Ch5 goes to 2000.

Ch 7 uses button input 0, down is off, up is on.  Requires same artificial PWM control.

Ch8, uses 2 button inputs.  "Down" turns Button 4 input on.  "Middle turns Button 3 and 4 off.  "Up" turns Button 3 on.

R-Lefebvre, 2015-03-16:
In case it's not clear, I want to use the Interlink controller to do functional testing via SITL, and wanted to get proper usage of Aux Function Switches (Ch7-8) which can't be properly done using Joystick buttons currently.

tridge, 2015-03-16:
as a workaround you can combine the interlink controller with rc commands in mavproxy to get the chan7/8 functions

R-Lefebvre, 2015-03-16:
Oh, you mean typing the RC commands into SITL manually?  Yes, I realized that option.  Was hoping to have the switches though, so I could change settings at about ~1Hz.

tridge, 2015-03-16:
http://dev.ardupilot.com/wiki/simulation-2/sitl-simulator-software-in-the-loop/using-sitl-for-ardupilot-testing/#using_a_joystick

meee1, 2015-03-22:
rob, can you post a screenshot of what it shows in MP?

meee1, 2015-03-22:
so to use this,
in the axis screen you need to pick custom0 or custom1.
in the button screen you need to pick button_axis, and pick the pwms you want to use for the button state.

R-Lefebvre, 2015-04-17:
Hi Michael, I tried to use this function unsuccessfully.  I just updated my MP by doing Check for BETA Updates.  I assume this change is in there?

Then I see for Ch7 and 8 I can select Custom 1 and Custom 2.  In the button section, there is a button_axis0 and button_axis1 selection.  But I couldn't seem to get this working.  The pwm settings don't seem to want to save properly.  I managed to get some response out of the radio calibration screen, but it's not working right.

Also, now my Interlink controller seems to have greatly reduced range.  The sticks only move PWM between about 1300-1600.  I'm not sure if this is related to this change or not.  This was working before.  I tried doing a Windows joystick driver calibration, but that didn't help.

Jman841, 2015-12-11:
The PWM settings still seem to be an issue. I am trying to set up a button to raise and lower the landing gear on a Quad from a joystick. Unfortunetly landing gear is not an option for buttons and I can't set the proper PWMs for one of the channels to control it via a Channel.

The other issue is, for all buttons there is only 1 function. it would be nice to be able to push a button once and it goes to X PWM, and when you push it again it goes to Y PWM for servo pass through so we can easily set more than one mode.