ArduPilot/MissionPlanner #817: max waypoints failure during loading WP File

buckker, 2015-03-17:
Hi

I find out that the message "to many waypoints" shows up when I load a WP File with over 256 numbers. 

If I create a mission with over 256 numbers, I can send it to the pixhawk without any problems. I can send the commands back to the computer. "Save WP File" works also, but to load the waypoints only works up to 256 numbers. 

I think there is restriction in the mission planner who cause this problem

Many thanks in advance

Michael

meee1, 2015-03-17:
please update to the latest MP, this was a known bug.

closed 2015-03-17 by meee1

buckker, 2015-03-18:
I uptated last Sunday to the latest beta version and it didn't work... I try today again and give you an update on this.

Thanks for your support

Michael