ArduPilot/MissionPlanner #819: Channel 6, 7, 8  function cant be assigned

chuti, 2015-03-25:
Since missionplanner 1.3.20 it is impossible to activate a function

Flashy-GER, 2015-03-25:
I can confirm this!
since MP 1.3.20 and above (Arducopter 3.2.1), the settings cant be activated.
Please have a look on this.
I did compass calibration with MP 1.3.21, the new values are not saved.

UAVHUB, 2015-03-26:
Confirmed here to MP 1.3.20

meee1, 2015-03-26:
fixed here 8fdc679f8864ccf4c437a719dfacce58e043f6b7

chuti, 2015-03-31:
Sorry, but it is not correctly fixed. It is impossible to store the values of Min and Max from channel 6 in the pixhawk.

meee1, 2015-03-31:
please provide more info, as it works fine for me. I can go from 0 to 10000

chuti, 2015-03-31:
If I set the values in the full parameter list it works. But if I want to set the Min and the Max value of channel6 in the Extendet Tuning it dont works right. I only can set values above siro. For example: If I set Min to 1,0000 and Max to 9,0000 it will work and in the setting of throttle accel D i can see a value between 0,001 to 0,009.
If I want to set the Min to 0,001 and the Max to 0,009 it dont works. 

i am sry for my bad English

meee1, 2015-03-31:
make sure you are using latest mp, there was a bug on older version.
latest is 1.3.23

chuti, 2015-03-31:
Thank you. I will check it out now. I tested 1.3.22

Flashy-GER, 2015-03-31:
@meee1 the latest 1.3.23 is not available via Update button.
so its possible to put values with 0.xx into the mask? 
Otherwise don't have a problem to put values > 0 into it, but it should be documented.

chuti, 2015-03-31:
I got 1.3.23 via the update button. There is no diffrent. Only integer numbers are possible. Then the values will appear as decimal numbers. Example:
Channel 6 = Throttle Accel kD
Min=1,000 (that works fine)
Min=0,001 (impossible to store)
Max=9,000 (that works fine)
Max=0,009 (impossible to store)
Throttle Accel D= a value between 0,001 and 0,009
I can live with it.

meee1, 2015-03-31:
please try  ,  and .

chuti, 2015-04-01:
Yes. I tried it. In my last post it is the result. But i can live with it.