ArduPilot/MissionPlanner #821: When BRD_SAFETYENABLE is set to 0 MP won't connect to Pixhawk

gmorph, 2015-03-27:
I set BRD_SAFETYENABLE to 0 in MAVProxy and then exited MAVProxy and power cycled the Pixhawk.  I started MAVProxy again and it was able to reconnect to the Pixhakw just fine - the safety switch is glowing solid red as expected.
I then disconnected the pixhawk and connected to my windows machine running a newly updated MissionPlanner and the connection timed out both when selection AUTO or specifying the COM port.
All connections to the pixhawk were done via USB.

jean-luc first reported this problem about 8th post down.
http://diydrones.com/group/arduboat-user-group/forum/topics/using-ardurover-for-a-4m-solar-powered-boat-for-long-distance?page=1&commentId=705844%3AComment%3A1945089&x=1#705844Comment1945089

meee1, 2015-03-27:
why is this a MP issue?

gmorph, 2015-03-27:
Hi Michael.  It only happens with MissionPlanner.  MAVProxy can connect to the Pixhawk just fine when BRD_SAFETYENABLE is set to 0 but MP doesn't seem to be able to.

meee1, 2015-03-28:
ive tested here, and cant find an issue. what firmware? does it happen with all firmwares? ie plane rover copter

closed 2015-03-31 by meee1