ArduPilot/MissionPlanner #823: Unable to upload a full mission's worth of waypoints 1.3.21 1.3.22

WickedShell, 2015-03-31:
Trying to upload a flight plan of 152 waypoints to the plane as far as mission planner is concerned it suceeded, however if you read back the waypoints it will have only succeeded in uploading between 60 and 100 waypoints. The dropped waypoints are scattered through out the mission, usually after 40 or so waypoints. 3DR radios were running SiK 1.9, most recent version of APM:Plane 3.2.3

The alarming aspect of this, is that it is a silent failure, and can leading to the plane navigating to unexpected locations.

The mission that didn't work is available here: http://pastebin.com/fJuetPLP

meee1, 2015-03-31:
please provide the tlog

WickedShell, 2015-03-31:
Sorry, the log can be found here: http://files.swiftradioplanes.com/2015-03-30%2016-26-26.tlog

meee1, 2015-03-31:
looks like I have found this issue, 
the issue only occurs in heavy packet loss situations.
ie via usb is not an issue, via radio can be

WickedShell, 2015-04-01:
1.3.23 fixed it, thanks!

closed 2015-04-01 by WickedShell