ArduPilot/MissionPlanner #824: Erle-Brain support

vmayoral, 2015-04-01:
Greetings,

@meee1 is it possible to add [Erle-Brain](http://erlerobotics.com/blog/product/erle-brain) (Linux autopilot) as a supported board to upload firmware through MP? What can we do to help?

meee1, 2015-04-01:
Hi, what would mp have to upload? and how would it interface to the BB?

vmayoral, 2015-04-01:
Hey @meee1,

Thanks for your answer. MP needs to upload solely the binary (e.g.: ArduCopter.elf) to a specific destination and overwrite the exisiting file. This destination is `/root/<binary-name>`. How can we help?

meee1, 2015-04-01:
ok, what is you current method of doing this? ssh? microsd physical? and the connection method you use? ie tcp,usb etc.

vmayoral, 2015-04-02:
@meee1 we generally `ssh` into the board, yes.

The connection method is TCP and we configure (be default) in our boards two connection mechanisms:
 - WiFi access (Erle-Brain creates its own WiFi network acting as a hotspot), Erle-Brain's WiFi interface IP address is `11.0.0.1`
 - USB-over-Ethernet, when pluging a mini-USB cable to the client USB port of Erle-Brain a network interface is created automatically in the brain. Erle-Brain's IP address over this interface is `192.168.7.2`

rmackay9, 2015-04-02:
I've posted an email to drones-discuss to try and clarify a standarize method for uploading components onto these linux boards.  All feedback welcome!

meee1, 2015-04-05:
https://groups.google.com/forum/#!topic/drones-discuss/u4KyOK0522s