ArduPilot/MissionPlanner #825: Survey Grid: Mission Planner is not populating the takeoff angle

jmachuca77, 2015-04-01:
When doing a takeoff with a plane the takeoff angle must be added to the takeoff waypoint. Currently only the target altitude is populated but the angle is left as 0. This results in a dive to the ground right after a hand launch. Please add a takeoff angle of 20 deg

[image]

Should be:
[image]

meee1, 2015-04-01:
Jaime, I do have code to do this already, where you connected to the plane when you programmed the mission?

jmachuca77, 2015-04-02:
No, I was not, and I did notice some of my missions do have it and some don’t. Is this only populated when you are connected to plane? Should it be always populated? I think copter will ignore it anyway right?

jmachuca77, 2015-04-02:
So this does seem related to #826 after I connect to a plane and do a mission the pitch angle is correctly populated.

meee1, 2015-04-05:
yes, correct

closed 2015-04-05 by meee1