ArduPilot/MissionPlanner #826: Flight Plan: Waypoint parameter descriptions for Takeoff missing

jmachuca77, 2015-04-01:
The waypoint description for takeoff angle is missing on the waypoint screen (first parameter "20" should read takeoff angle in description):

[image]

meee1, 2015-04-01:
this goes with #825, MP remembers the last device type you where connected to. and the last device time you where connected to was a copter, not a plane.

jmachuca77, 2015-04-02:
Verified, after connecting to a plane the description appears correctly. Should this just stay the same on both plane and copter?

meee1, 2015-04-02:
no, because there are several commands that differ, that's why they are like that.

if you click just under the baudrate selection box another drop down will appear, where you can override the current type

closed 2015-04-05 by meee1