ArduPilot/MissionPlanner #829: Waypoints not loading correctly and old waypoints showing up in middle of mssion.

renoflash, 2015-04-02:
Today I tried to send waypoints a survey grid using distance for trigger. Mission planner would act like the wp were sent but when I would read back, old waypoints from a previous mission would be intermingled. I am using Mission Planner v. 1.3.23 and Arducopter 3.2.1. I tried this with two separate computers both running windows 8.1. Mission planner was updated yesterday from 1.3.20 on one machine and 1.3.16 on the other. I decided to uninstall mission planner and reinstall 1.3.23 and it seems to work. There may be fly aways if people don't check the waypoints as mine started to fly to an old waypoint after waypoint 5.
I have telemetry file if you are interested.

meee1, 2015-04-03:
please provide the tlog

meee1, 2015-04-03:
there was a potential issue with 1.3.20 when using 3dr radios, usb was fine though.

renoflash, 2015-04-03:
I was using clone 3dr radios. I thought about using usb. As soon as I get home I'll send my tlog file and the way point file. I have not uninstalled mission planner on my laptop yet so I should be able to recreate issue if you need me to.

Sent on a Sprint Samsung Galaxy Note® 3

<div>-------- Original message --------</div><div>From: Michael Oborne <notifications@github.com> </div><div>Date:04/02/2015  7:28 PM  (GMT-08:00) </div><div>To: diydrones/MissionPlanner <MissionPlanner@noreply.github.com> </div><div>Cc: renoflash <robdunbar@outlook.com> </div><div>Subject: Re: [MissionPlanner] Waypoints not loading correctly and old waypoints showing up in middle of mssion. (#829) </div><div>
</div>
there was a potential issue with 1.3.20 when using 3dr radios, usb was fine though.

---
https://github.com/diydrones/MissionPlanner/issues/829#issuecomment-89126778

renoflash, 2015-04-03:
What is the best way to upload the Tlog file? I haven't found a way to post it using GitHub. I tried responding via email but it got kicked back.

meee1, 2015-04-04:
google drive, one drive, dropbox

renoflash, 2015-04-04:
Here is the link to the telemetry file and the waypoint files. I was trying to take pictures using CHDK and a trigger setup. I started with using waypoints for the trigger points but I got errors when uploading the waypoints so I switched to distance to trigger waypoints. So you will notice I loaded some waypoints a few times. The waypoints show up correctly when I finally do get up and fly but you will notice the copter starts heading away.
https://onedrive.live.com/redir?resid=acb3e97cb018e013!9682&authkey=!ABmCXav-JZIpT28&ithint=folder%2ctlog

meee1, 2015-04-05:
did you have more than on gcs connected?

renoflash, 2015-04-05:
Yes. After I was having troubles I uploading one way point using Tower and a bluetooth bridge.I then retried Mission Planner still with no success.

meee1, 2015-04-05:
there was an issue in mp with uploading wp's over a lossy link, but that should have been fixed in 1.3.23

but I think what makes it more odd is that mp and DP look to have been requesting wp's at the same time, make it very messy.

renoflash, 2015-04-05:
The odd thing is I was only using mission planner at first. I would upload the waypoints and then download them to check. What came back was some of the old points from a previous mission. I thought maybe using droidplanner would clear up the waypoints in pixhawk. There was a short time when I had both MP and DP running by accident. I turned off DP radio and still had issue. Tried powered off/on on reset on pixhawk and still had the issue. Then when I got home I tried the same thing on a different computer but had the same problem. With a fresh install of MP the upload and download of waypoints worked fine. 
When I get back home tomorrow, I'll try again on the one computer I haven't done a fresh install and just see if the issue reoccurs. I thought there might be an issue but if no one else has had this come up then its something I did.

meee1, 2015-04-05:
can you test with the latest 1.3.23? make sure you redownload it, just to be 100% sure.

renoflash, 2015-04-06:
Ok I tried this time with the fresh install of mission planner. It is still acting up. Here is a link to the TLOG file and a video screen capture of me creating the waypoints from a polygon https://onedrive.live.com/redir?resid=acb3e97cb018e013!9827&authkey=!AFJPy6HiTpdPA_U&ithint=folder%2ctlog . I create the waypoints using Survey and upload them. I then download the points and you will see some old waypoints. This was done using radios. Same thing happened over USB. I even tried AC 3.2 and same thing. I also tried this on a hobbyking pixhawk clone with v. 1.3.23 and AC 3.2.1 and same thing occurs. I was beginning to wonder if I had some corrupted eeprom.

I did get it to work when I downgraded to version 1.3.16. However, when I tried the do-digicam-control and there was 280 waypoints it would not load all the waypoints. There appears to be a 149 waypoint limit. I thought it was up in the 700s now.

meee1, 2015-04-06:
what is the build number on your mp? should be in the title bar

renoflash, 2015-04-06:
build no. 1.1.5568.40256

meee1, 2015-04-06:
I will be releasing a new mp version with this fix.

renoflash, 2015-04-06:
Thanks for your hard work. I put some in the tip jar.
Rob

renoflash, 2015-04-06:
Thanks Michael.
I just tried it using 1.3.24 Build # 1.1.5574.19016 and it is working great. I've tried two different locations to set waypoints. I even uploaded 405 waypoints and checked the download and they all came back right were they should be.

closed 2015-04-06 by renoflash