ArduPilot/MissionPlanner #831: MP 1.3.23 - Extended Tuning PID Values Incorrect Decimal Place

Hands0n, 2015-04-04:
In Mission Planner 1.3.23, Extended Tuning there are two apparent issues relating to the Rate/Roll PID where the values are all now displayed to the right of the decimal point.   This makes it confusing to use coming from earlier versions (I came from 1.3.19 directly to 1.3.23, so this may have changed in .22 even).  

Default Rate/Roll PID are now displayed as P = 0.150 | I = 0.100 | D = 0.004

I have also observed that the maximum values of PID are also fixed as follows; P = 0.200 | I = 0.500 | D = 0.0020

Was this intentional or is this a genuine bug?

See attached screen shots of the above.

Default Rate/Roll Values in 1.3.23
[image]

Max Rate/Roll Values in 1.3.23
[image]

Hands0n, 2015-04-04:
Attached are the Extended Tuning panel screen shot from an earlier version of MP (from memory MP 1.3.19) 

Observe that Rate/Roll P is 0.2630 which could not be set in the incumbent MP 1.3.23

[image]

meee1, 2015-04-04:
mp now inforces the mins and maxs as defined here 
https://github.com/diydrones/ardupilot/blob/master/ArduCopter/Parameters.pde#L590
using the raw param screen you can put in any value you want still however

Hands0n, 2015-04-04:
Hi Malcolm

I've just had a reply on github.  Have a look.  The code snippet he points
to is as follows:

// PID controller //--------------- // @Param: RATE_RLL_P // @DisplayName:
Roll axis rate controller P gain // @Description: Roll axis rate controller
P gain. Converts the difference between desired roll rate and actual roll
rate into a motor speed output // @Range: 0.08 0.25 // @Increment: 0.005 //
@User: Standard // @Param: RATE_RLL_I // @DisplayName: Roll axis rate
controller I gain // @Description: Roll axis rate controller I gain.
Corrects long-term difference in desired roll rate vs actual roll rate //
@Range: 0.01 0.5 // @Increment: 0.01 // @User: Standard // @Param:
RATE_RLL_IMAX // @DisplayName: Roll axis rate controller I gain maximum //
@Description: Roll axis rate controller I gain maximum. Constrains the
maximum motor output that the I gain will output // @Range: 0 4500 //
@Increment: 10 // @Units: Percent*10 // @User: Standard // @Param:
RATE_RLL_D // @DisplayName: Roll axis rate controller D gain //
@Description: Roll axis rate controller D gain. Compensates for short-term
change in desired roll rate vs actual roll rate // @Range: 0.001 0.02 //
@Increment: 0.001 // @User: Standard

Hands0n, 2015-04-04:
Thank you Michael. I think I understand what has been done.  Was there a reason for such constraints?   Understand that broader ranges can be set in raw param screen. But it takes away from the friendly functionality of the Extended Tuning panel. 

Will make sure to pass this advice on if asked.  
Thank you.

meee1, 2015-04-05:
the range will be limited to safe numbers, but depending on your frame/motor setup this will change. and your numbers may vary, as they do it seems.

closed 2015-04-05 by meee1

rmackay9, 2015-04-08:
Sorry if you're getting the run-around here but all we can do in arducopter is increase the allowable range of the parameter values. So for example, it looks like 0.25 is the maximum for the Roll/Pitch Rate P. We could increase that to say 0.28. We can't do all the other things mentioned like turn-off range checking min MP by making an ardupilot/copter change.

meee1, 2015-04-08:
all I was suggesting was to increase the max, as the current max doesn't cater for this frame type. I was trying not to modify mp because it could lead to people going way beyond the normal range.

Hands0n, 2015-04-08:
Hi guys, thanks for commenting back. 

Just to flesh out the request requirement basis.

So my frame type is extremely rigid, there is zero flex in the frame at all.  We have seen the same issue in using the Extended Tuning panel for three of this construction. It is aluminium arms based on a rigid core - the entire emphasis of the design is the rigidity of the frame. On a more wobbly frame, such as TBS Discovery or Firefly 450 (DJI 450 clone) with the plastic arms the APM PIDs fit the constraints introduced into the Extended Tuning panel. 

Motors are DJI 2212 920KV with 8045 plastic props.  Vibration is almost a flat line, substantially within "Acceptable" as defined in the ArduPilot wiki. 

When you refer to "normal range" what does that really mean if I can bypass by going to Full Parameters? I am not certain what the previous upper limit is but it evidently was sufficient to permit the 0.2630 value for Rate Roll/Pitch.  That would be "normal" for the frame type being used here.  By constraining the value range surely it limits the frame type that APM/Pixhawk will support in the future?  Is that really the intention?

I would really appreciate and encourage a solution as I do feel that it detracts from the entire project by effectively taking Extended Tuning away from anyone who uses a frame that does not fit within the "normal" constraints.  It makes the setting up and tuning of ArduCopter a significantly different and inferior proposition than before. We cannot even use the channel 6 dial to perform in-flight tuning because of this.

ayoubzar, 2016-08-19:
Hi guys 
please I having trouble to adjust my PIDs values for my aircraft (Volantex RC 757-3)
I am using STIL with MP planner after I made a module of it in plane maker .
But it seems like doesn't respond to the parameters 
is there any way to calculate them or I should go out  and flight it manually ?
I want to be sure via simulation first is that will work or I am just wasting my time ?
please any tips will help

Hotwire-, 2017-07-24:
Was there ever a solution to this problem. Extended tuning is not functional and I suspect its becaus the default values file, is missing or not being changed. If I knew the file name I could edit it?

mirikle, 2017-12-03:
I have an "overpowered" setup, currently the P gain for ROLL and PITCH doesn't  allow me to go below 0.08, however there's still some twitchiness given this P, but already flyable.

Wonder if there's a way to set the P for ROLL and PITCH below 0.08? I searched around and couldn't find why the minimum value is set to 0.08. Thanks!