ArduPilot/MissionPlanner #832: Feature Request: Plot ground station GPS and display distance from the aircraft to the ground station

WickedShell, 2015-04-05:
The hardware we use as a GCS has an internal compass that publishes a standard NMEA stream on a COM port, it would be really nice to be able to plot where the GCS currently is on the map, as well as the distance from the GCS to the last reported aircraft position.

I think most of the work is already covered with Follow Me mode, but I don't see a way to use Follow Me to plot the current GCS position.

meee1, 2015-04-05:
give moving base a shot.
control-f > moving base

WickedShell, 2015-04-05:
That sounds like what I'm looking for, although in testing it seems to crash everytime, since I can't copy paste the error I've attatched a screenshot of it. (sorry about the werid colors, they are selected for working well on a transflective screen in the sun)
[image]

meee1, 2015-04-05:
2 options
user run as administrator.
or
on the link you use to start mp, set the working directory to the mp directory

closed 2015-04-21 by meee1