ArduPilot/MissionPlanner #834: Feature Request: In Extended Tuning Permit Advanced User Mode

Hands0n, 2015-04-07:
I am raising this in reference to an issue I raised earlier and which is now closed, see https://github.com/diydrones/MissionPlanner/issues/831

Resorting to Full Parameters as a solution takes away from the convenience and functionality of MP's Extended Tuning panel.   

I would like to suggest a feature in Extended Tuning that allows "Advanced Use" whereby when enabled the user is not constrained to ranges of values possible.  The "Advanced Use" switch would require a deliberate action on the part of the person using Mission Planner.  By default it would be "Off" (ie. constrained PID ranges).

Why?  I use ArduCopter in four different aircraft and in at least two of them the constrained ranges are not suitable.  Mission Planner, as a tool, is wonderfully comprehensive and flexible.  Other panels allow Advanced Use, so I would appeal for this to be provided in a future version of MP.

meee1, 2015-04-07:
this needs to be on the ardupilot issue list, as the limit is defined from there.

Hands0n, 2015-04-08:
Thank you. I will raise it there.

closed 2015-04-21 by meee1