ArduPilot/MissionPlanner #835: SITL and XPlane simulation

smr277, 2015-04-07:
The XPlane simulation works with HIL, however when I try to connect using the "SITL" checkbox, mission planner doesn't connect to the SITL. I've checked all the ports and IP addresses and I can connect to the SITL with MAVProxy and QGroundControl. I can even connect to the SITL with the "Connect" button and selection of "TCP" in Mission Planner, however this doesn't send the data from XPlane back to the SITL.

meee1, 2015-04-07:
did you check the advanced settings?

smr277, 2015-04-08:
Yep. The problem seems to be that the SITL uses two UDP ports where MissionPlanner uses a single TCP port

kjeremy, 2015-10-02:
I'm seeing the same issue.

ayoubzar, 2016-08-09:
Is there any solution found for that ? I am facing the same problem 
please

magicrub, 2016-08-09:
Try udp

ayoubzar, 2016-08-09:
Is there is any requirement for the direction of software (Xplane and MP should be in the same direction ?)
because i tried UDP and TCP  as well ,always MP said heartbeat packet missed 
any tips will be very appreciate it 
I spent like two weeks trying to figure it out without any result

magicrub, 2016-08-09:
Do you have the normal SITL sim working?

ayoubzar, 2016-08-09:
Yes it is working normally but when i want to contact it with mission planner a message pops up said heartbeat packet missed .
all the ports are good ,IP as well ,more than that sometimes it works but most the time is not 
what will the problem in this case Sir please

ayoubzar, 2016-08-10:
Sir is this software need any extra codes for the communication or all includes in them ?
what is the relation of Mavlink with SITL ?
as far as I understand heartbeat packet send by Mavlink verifies is the APM connected or not but we talking about SITL here there is no Hardware .
why this message pops up ?
please any help I am like loser now

Abdullah1990, 2016-11-16:
I am facing the same issue. As I would like to use the SITL with x-plane but mission planner will not connect with the Arduplane.exe via TCP when I have chosen x-plane as a simulation platform. I think the issue is with in the arduplane code.

meee1, 2016-11-16:
are you using the latest version of mission planner? have you allowed missionplanner, SITL and Xplanes through the firewall?

Abdullah1990, 2016-11-16:
Yes I am using the latest mission planner 1.3.41. And I did disable the windows firewall entirely. But i think the STIL is not sending any messages on tcp port 5760

nocternal, 2016-12-09:
I also use win10 and try wiki‘s three ways：Using SITL from MissionPlanne；Using SITL with your own GCS；Using SITL with sim_vehicle.py
None of them worked...no matter use Mavproxy or MP，they can't get TCP or UDP messages...

t4r4n15, 2017-01-19:
Same as @nocternal I've tried every way described in documentation, none works.

mojtabacazi, 2018-04-10:
3years later, and I have the same problem. Any updates?

nocternal, 2018-04-10:
No... I finally use Flightgear instead of XPlane to have visualization of SITL. At least Flightgear works

mojtabacazi, 2018-04-10:
@nocternal enabled xplane data outputs and it started working. http://ardupilot.org/dev/_images/mavlinkhil1.jpg