ArduPilot/MissionPlanner #839: CH6 Roll/Pitch gain Max-Min not written correctly

jmachuca77, 2015-04-10:
This was reported on one of the Facebook groups:

"The latest Mission Planner appears to have a bug in it!

I setup ch6 to adjust roll/pitch p-gain. I entered the desired values of 3.5-6.5 and write data back. When i checked, the PID values were set to 0.0035-0.0065!

Those values would crash your machine!"

[image]

jmachuca77, 2015-04-10:
I confirmed the error with SITL
Set CH6 on mission planner extended running to CH6 Rate Roll/Pitch kP, Min 3.000 Max 6.000
when reading back parameters the Rate Roll and Rate Pitch P parameters are actually set to 0.003 - 0.006 range, they should be set to a range of 3.0 - 6.0. 
Not sure where the error is here. Might be in Mission planner sending the wrong parameter range, or could be on Copter.
[image]

meee1, 2015-04-20:
so should be a factor of 1000 in there

closed 2015-04-20 by meee1