ArduPilot/MissionPlanner #840: Beat Firmware link should show Pixhawk's version by default

rmackay9, 2015-04-12:
Currently the Initial Setup > Install Firmware page's Beta Firmware link shows the APM version by default but with the move to the higher CPU boards (i.e. Pixhawk) and dropping support for the APM2.x board, it would be good to show the Pixhawk versions by default.

This could apply to the software versions for official releases as well although ideally for the official versions it would show the version appropriate for the user's board (although it may be difficult to know which board is attached at the moment that the versions are downloaded from the central server)

meee1, 2015-04-12:
https://github.com/diydrones/MissionPlanner/commit/e1985497e2be27ca7017b0e118252b69310f2055

closed 2015-04-20 by meee1

rmackay9, 2015-04-20:
thanks a lot for that.