ArduPilot/MissionPlanner #849: Live Calibration fails with unexpected error midway through calibration (reference to User hog?)

visser, 2015-04-25:
Hi diydrones, here's the exception details (summary and screenshot) and my config.xml, Windows 8.1, all updates with 3DR Pixhawk and NEO-6M gps+compass, MP: Mission Planner 1.3.25 build 1.1.5589.39444 (official)

[image]
[image]

Object reference not set to an instance of an object.

at MissionPlanner.MagCalib.prd_DoWork(Object sender, ProgressWorkerEventArgs e, Object passdata) in C:\Users\hog\Documents\Visual Studio 2010\Projects\ArudpilotMega\ArdupilotMega\MagCalib.cs:line 400

visser, 2015-04-25:
Re-installed Mission Planner from the Help screen and was able to complete calibration :)

closed 2015-04-25 by visser

meee1, 2015-04-26:
one question, how did you install mp in the first instance? zip? msi? etc,

visser, 2015-04-26:
Always via MSI installed and once installed then using the auto updater
within MP :)