ArduPilot/MissionPlanner #850: Elevation Data preflight check

Jman841, 2015-04-25:
Is it possible to add a preflight check button in the Flight Plan to give a warning if any part of the flight is below the elevation data? 

Recently we had an issue where absolute altitude was accidentally used or somehow sent as the elevation data for a flight plan when relative should have been used. Luckily it did not result in a crash as the behavior was noticed and switched to manual mode, however, is it possible to add a preflight check or warning if mission planner notices something like this? Also, maybe any other warnings for things that could result in mission failure?

Thank you

meee1, 2015-05-04:
copter or plane?

Jman841, 2015-05-04:
This was on ArduPlane, although the preflight check button could benefit the copter firmware as well

Jman841, 2015-05-04:
Some things it could check, elevation graph data vs. flight plan, minimum speed requested during flight plan vs. stall values, maximum pitch angle changes, landing pattern pitch changes, and any other critical flight plan settings that could result in mission failure if not confirmed by the user.