ArduPilot/MissionPlanner #853: An idea of adding all-in-one Safety Status Display

SkiTiSu, 2015-04-28:
> Cloned from diydrones/ardupilot#2171

At present, the Pre-Arm safety check only return errors via text messages every 30 sec or arming. We could only receive one error every time.

I'd like to add a new mavlink message to represent the safety status in 1Hz, including all checks' results. Like this:

```xml
<enum name="SAFETY_STATUS_FUNC">
    <description>For SAFETY_STATUS.</description>
    <entry value="1" name="PREARM_STATUS_FUNC_RC">
        <description>RC not calibrated</description>
    </entry>
    <entry value="2" name="PREARM_STATUS_FUNC_BARO">
        <description>Baro not healthy</description>
    </entry>
    <entry value="4" name="PREARM_STATUS_FUNC_ALT">
        <description>Alt disparity</description>
    </entry>
    <entry value="8" name="PREARM_STATUS_FUNC_COMPASS_HEALTH">
        <description>Compass not healthy</description>
    </entry>
    <entry value="16" name="PREARM_STATUS_FUNC_COMPASS_CALIBRATED">
        <description>Compass not calibrated</description>
    </entry>
    <entry value="32" name="PREARM_STATUS_FUNC_COMPASS_OFFSETS">
        <description>Compass offsets too high</description>
    </entry>
    <entry value="64" name="PREARM_STATUS_FUNC_MAG_FIELD">
        <description>Check mag field</description>
    </entry>
    <entry value="128" name="PREARM_STATUS_FUNC_COMPASS_INCOSISTENT">
        <description>compasses inconsistent</description>
    </entry>
    <entry value="256" name="PREARM_STATUS_FUNC_FENCE">
        <description>check fence</description>
    </entry>
    <entry value="512" name="PREARM_STATUS_FUNC_INS">
        <description>INS not calibrated</description>
    </entry>
    <entry value="1024" name="PREARM_STATUS_FUNC_ACCELS_HEALTHY">
        <description>Accels not healthy</description>
    </entry>
    <entry value="2048" name="PREARM_STATUS_FUNC_ACCELS_INCONSISTENT">
        <description>Accels inconsistent</description>
    </entry>
    <entry value="4096" name="PREARM_STATUS_FUNC_GYRO_HEALTHY">
        <description>Gyros not healthy</description>
    </entry>
    <entry value="8192" name="PREARM_STATUS_FUNC_GYRO_CAL">
        <description>Gyro cal failed</description>
    </entry>
    <entry value="16384" name="PREARM_STATUS_FUNC_GYRO_INCONSISTENT">
        <description>Gyros inconsistent</description>
    </entry>
    <entry value="32768" name="PREARM_STATUS_FUNC_BOARD_VOLTAGE">
        <description>Check Board Voltage</description>
    </entry>
    and more....
</enum>

<message id="226" name="SAFETY_STATUS">
   <description>Add soon.</description>
   <field type="uint32_t" name="prearm">Add soon.</field>
</message> 
```

And in ```static bool pre_arm_checks(bool display_failure)```, change them like this

```cpp
pre_arm_rc_checks();
    if(!ap.pre_arm_rc_check) {
//      if (display_failure) {
//          gcs_send_text_P(SEVERITY_HIGH,PSTR("PreArm: RC not calibrated"));
        safetystatus += SAFETY_STATUS_FUNC.PREARM_STATUS_FUNC_RC;
        }
//      return false;
        flag = false;
    }
```

In MP, add a UserControl like this:
[image]

May it affect the performence, or have other bad effects?

Thanks.

meee1, 2015-04-30:
my thought would be to add a tool bar strip (next to donate button), in the main menu, with colour indicators, and appropriate mouse over info.

SkiTiSu, 2015-05-02:
@meee1 Thank you for your suggestions~

meee1, 2015-05-04:
this should tie with #851