ArduPilot/MissionPlanner #856: Geo Ref tool CAM message validation

nicksargeant, 2015-04-29:
Minor enhancement: While the CAM message syncro should either work or fail with a warning, I would find it useful for the log window to display the calculated offset time for a means of validation as shown:
[image]
As they should all return the same value, being able to quickly view the actual time offset allows easier verification that georeferencing has worked successfully over multiple flights that have to be individually georeferenced.
Thank you as awlays!

meee1, 2015-04-29:
could you provide an example? just so I fully understand what you are requesting?
we have the cam message time, and we will have the image time, is that what you want? the delta time?

nicksargeant, 2015-04-30:
Sorry Michael, my request wasn't particularly clear. As you discerned, all i would like the tool to do is somewhere display the delta time calculated when using the CAM message to georeference images.

closed 2015-05-04 by meee1