ArduPilot/MissionPlanner #857: "trigger camera now" does not work with smart camera

jmachuca77, 2015-04-29:
With the addition of support for mavlink aware camera modules, the "trigger camera now" does not work with these module. In order to trigger these cameras, a command long message with a MAV_CMD id of 203 (MAV_CMD_DO_DIGICAM_CONTROL) should be sent. Param5 on the message must not be equal to 0 in order to trigger the camera. Also the target system should be correctly set up and the component id should be set to 0 in order to broadcast to all components.

jmachuca77, 2015-04-29:
it seems currently at least ArduCopter does not support triggering the camera from the command long, if it is connected directly to the autopilot. Discussion on what message to support (ardupilot specific vs. common mavlink message) to follow on Drones Discuss.

meee1, 2015-04-29:
mp sends a DIGICAM_CONTROL packet
with the shot field = 1;

meee1, 2015-04-29:
however you are expecting a COMMAND packet it would seem

meee1, 2015-04-29:
ie 
#if CAMERA == ENABLED
    case MAVLINK_MSG_ID_DIGICAM_CONFIGURE:      // MAV ID: 202
        break;

    case MAVLINK_MSG_ID_DIGICAM_CONTROL:
        camera.control_msg(msg);
        log_picture();
        break;
#endif // CAMERA == ENABLED

taedla01, 2015-05-02:
Hello everyone,
Just my two cents...
For me, with Arducopter code, when I use "trigger camera now" the camera is triggered, but not CAM dataflash message seems to be created on MicroSD. With Arduplane it works like a charm... Not sure if it is a problema concerning Mission Planner or Arducopter code... 

Cheers,
Dario.

meee1, 2015-06-03:
Jaime can i close this?

jmachuca77, 2015-06-03:
We still need a change. We need a decision as to what message to use. Randy and me both feel we should use the common message and not the ardupilot specific message. It would just be nice to know what everyone else thinks.

jmachuca77, 2018-08-16:
This works, how ever there is a double trigger when using a companion computer as ardupilot forwards the received message from the GCS to the companion, but also generates its own trigger message therefore triggering the camera twice. I opened an issue in Ardupilot, closing this one. Issue is ardupilot/ardupilot#7159

closed 2018-08-16 by jmachuca77