ArduPilot/MissionPlanner #858: Tracker

ryanjohnston4, 2015-05-01:


meee1, 2015-05-01:
do you mean for the internal mp tracking? or the hardware based antenna tracker?

ryanjohnston4, 2015-05-01:
The hardware based antenna tracker - to have an antenna point back to the home or set location from the air.

meee1, 2015-05-01:
ok, so its not the hardware tracker at all?
your talking about using a gymbal on a plane to point at home?

ryanjohnston4, 2015-05-01:
I mean loading up the antenna tracker firmware on a separate onboard APM (independent of the flight controller unit) and using it on a small gimbal to point the antenna home. Directional antennas in 5.8 spectrum are so small with high gain; would be extremely beneficial for link quality.

rmackay9, 2015-05-01:
Ok so this request probably belong in the ardupilot issues list then.  You want to add it there?
https://github.com/diydrones/ardupilot/issues

I've been thinking of adding this to the antenna tracker firmware actually.

closed 2015-05-01 by rmackay9

ryanjohnston4, 2015-05-01:
Thanks will do now.   

Would be a really great function!