ArduPilot/MissionPlanner #859: Optional Hardware: add support for a second battery monitor

taedla01, 2015-05-02:
Hello everyone,
Now that Arduplane and Arducopter code supports the use of a second battery monitor through the Pixhawk ADC ports, would be great to configure it like the main battery monitor, i.e, through the optional hardware tab instead of the full parameter list.

Thanks for your time and your effort,
Regards from Spain,
Dario.

proficnc, 2015-05-02:
+1 pixhawk 2 will have capability for 2 power modules, so it would be great to be able to display the second battery.

lvale, 2016-02-15:
+1 Second that

taedla01, 2016-02-17:
Hello everyone,

This issue has been solved since Mission Planner 1.3.33 has been released so I close the issue. Take a look at the Pixhawk pinout adc references for know more about the second battery monitor implementation.

Cheers,
Dario.

closed 2016-02-17 by taedla01

lvale, 2016-02-17:
Although it can be configured on the PixHawk, unless you add the fields manually there is no feedback. Case of the HUD, the rede letters I assume are Batt1 only.

[image]

[image]

[image]