ArduPilot/MissionPlanner #861: Telemetry module setup part is broken in latest MP Beta

nhadrian, 2015-05-04:
Hi, I noticed that in the latest Beta MP firmwares the Telemetry module setup page is broken, ends up with some error messages. My current version is build 1.3.25.2 build 1.1.5599.32514 (but the problem exists till the latest 4-5 betas). I attached a sample of my error message, sorry about the language, I'm using hungarian Windows.

Regards
Adrian
[image]

meee1, 2015-05-04:
I just tested ant it seems to work fine.

what frequency radios are you using?
and what firmware version?

nhadrian, 2015-05-04:
I'm using MP (Multipoint) SiK 2.4 on 868 MHz HM-TRP module.
Before error message, the firmware version just appears, so it looks Mission Planner can read something. 
BTW, I can handle any settings through AT commands, but it would be much easier through Mission Planner...

Also, I had a suggestion earlier, when MP firmware is detected, additional parameters should appear (I'm not sure but maybe S15 is refered to something else in normal SiK firmware...):
-    S15: NODEID
-S16: NODEDESTINATION
-S17: SYNCANY
-S18: NODECOUNT

Mode detailed description can be found here:
https://github.com/RFDesign/SiK/tree/SiK_Multipoint

meee1, 2015-05-04:
can you download the standalone config tool and run "ATI3" on the radio, and see what it returns?

meee1, 2015-05-04:
https://github.com/RFDesign/SiK/blob/8fc7d46d494ea8e977aec42af503bdbcf9356f36/Firmware/include/board_info.h#L44
this is the official list

nhadrian, 2015-05-04:
Yes, of course. It it enough to check through terminal? If not, please sen me a link to the latest config tool. Thanks in advance.

BTW, 2-3 months earlier (I don't remember exactly) the config tool in MissionPlanner worked great with Multipoint firmware, except S15...S18 (described above).

meee1, 2015-05-04:
terminal is fine, so long as you know how to get into it

meee1, 2015-05-04:
the error listed in your screenshot is at the frequency detection stage, that's why im asking about it

nhadrian, 2015-05-04:
yep, I'll check in the afternoon and leave a comment.

Till then, one interresting for me is that the folder listed at the end of the message 'c:\users\hog\...' doesn't exist on my PC.

nhadrian, 2015-05-04:
OK, I tested with ATI3. 
A got "134" as a reply.

nhadrian, 2015-05-11:
Any updates on this issue? Is there anything else I can help you identify the problem?

meee1, 2015-05-11:
where did you get the firmware you are running on your radios?

nhadrian, 2015-05-11:
I got a compiled version from the MP (Multipoint) SiK firmware's author.
The only information I have is that the firmware is MP SiK 2.4.

Please note that some months ago I could use the inbuilt configurator for my radio modules, and the firmware on them is still the same.

That's why I think the change  is related to Mission Planner.

nhadrian, 2015-05-11:
I tested with the 1.3.2 3DR config tool, it recognizes all settings properly. Screenshot attached.
[image]

nhadrian, 2015-05-21:
Hi, is there any updates on this issue? 
Thanks in advance!
Regards, Adrian

meee1, 2015-05-21:
can you post you missionplanner.log file from when you do your test.
also possibly the firmware you uploaded to the radios as well