ArduPilot/MissionPlanner #864: Incorrect body type on map

TerryPts, 2015-05-10:
I don't use mission planner all that often so I don't know how new this is but I think the vehicle icon on the map in the latest release is wrong. I have an X and it is showing up as a +. I know it used to show correctly and I have gone through the wizard to be sure the correct style is selected so I'm not sure if it's a MP display issue or if the copter has somehow been configured as a +.

jmachuca77, 2015-05-10:
If you are referring to the vehicle icon on the map, that is only a representation I believe it's the same for all copter types.

Best regards,
Jaime Machuca

TerryPts, 2015-05-10:
It may be now but at one time it was in the X configuration, that's why I noticed it. As long as I know MP hasn't flipped configs on me then I'm not concerned.