ArduPilot/MissionPlanner #865: Beta Firmware descriptions incorrect on Install Firmware page

ggregory8, 2015-05-11:
When selecting the 'Beta firmwares' link on the Install Firmware page it retrieves the FW list and changes the descriptions, but they are incorrect and showing V3.2 instead of V3.3-rc3, at least for AC. You can see this and the MP version in the screenshot below. Running on Win8 x64
[image]

meee1, 2015-05-11:
I just tested, and it worked fine. please try it again.

ggregory8, 2015-05-11:
Still the same issue. I noticed this on the older version on MP as well. Maybe it is something specific to my setup.

As mentioned it is only the descriptions that are wrong. It still loads AC3.3-rc3

meee1, 2015-05-11:
please upload your missionplanner.log

meee1, 2015-05-11:
and do you use a proxy server?

ggregory8, 2015-05-11:
Log found here. Last thing I done was opened MP. Went to install firmware page, then clicked beta firmware and ok on prompt box. Then closed.
https://drive.google.com/file/d/0B2tRJoqjaZmxQi1GaHk2ZGp5M28/view?usp=sharing

No proxy.

meee1, 2015-05-11:
please update your mission planner version!!
the firmware version its looking up for for the apm2, which is no longer supported, and explans the older version label

closed 2015-05-11 by meee1

ggregory8, 2015-05-12:
Sorry I thought I was running the latest version of MP as it normally asks to update if not. Just discovered the update option in Help tab.
Thanks that has fixed it.