ArduPilot/MissionPlanner #866: Failsafe when Rx is set to "no pulses"

mpjf01, 2015-05-11:
There is a FS option discussed in the Wiki which is activated by the radio sending no pulses. The Wiki suggests that in this case the throttle value is irrelevant, yet if it is set too high there is a pre-arm warning.

MP does not offer explicitly a "no Pulses" FS option, it obviously fits into the Radio FS subsection but there is no distinction made between it and the Throttle FS option, leading the user to believe that a throttle PWM is relevant, reinforced by the Pre-arm error. 

It would be helpful if the user interface for this section could be updated and the error message sorted out.

meee1, 2015-05-11:
ca you please provide the wiki page

mpjf01, 2015-05-11:
http://copter.ardupilot.com/wiki/configuration/throttle-failsafe/#set-up_for_no-signal_method

rmackay9, 2015-05-11:
the difficulty is that the copter radio failsafe will trigger when either the throttle value goes too low or it stops receiving pulses.  Some receivers could make use of both methods.  For example a futaba receiver is normally set-up to pull the throttle low but if the receiver loses power completely then it would trigger the no-pulse radio checks.
I think it might be best to simply fix this in the documentation.

rmackay9, 2015-05-11:
I've modified the wiki page to say:
Because the throttle is not pulled low, there is normally no need to adjust the “FS Pwm” value from it’s default (975).  Just ensure that it’s well below (i.e. at least 10pwm points) below the minimum value that channel 3 (throttle) can be.

mpjf01, 2015-05-11:
Presumably all receivers will, in the event of power loss, send no pulses. So presumably there has to be a "default" FS setting that detects that situation and triggers the desired action. Perhaps the developers believe that this is implicit/obvious and doesn't need to be documented anywhere. The forums, however, suggest that this is not universally recognised initially, although readily understood when explained.

The MP UI explicitly deals only with the throttle FS option. In the case of receivers that can be set to emit no pulses as a FS indicator there is nothing in MP that indicates what the user is to do. The presence of a mandatory throttle value suggests that it is needed for some reason when in fact it is not. Why not just have a tick box asking the user to explicitly state that he wants a no pulses option only and grey out the throttle PWM so it's clear that it is not needed? 

And add something that allows the user to know that he is selecting both a no pulses and a throttle FS option for receivers that require both.

iskess, 2015-05-13:
Does the "no pulse FS" work with APM:Plane yet? I read somewhere it was copter only.