ArduPilot/MissionPlanner #867: add param tag of @RebootRequired

magicrub, 2015-05-12:
Refer to https://github.com/diydrones/ardupilot/issues/2278

Adding a new param description that notifies the GCS that a reboot is needed for a change on a given param is needed. "@RebootRequired:false" is optional so not all params will have this tag.

Suggested behavior:
If pushing any param(s) that have @RebootRequired: true then notify the user once that a reboot is required.

magicrub, 2015-05-12:
the terminology may be changing, track the issue in https://github.com/diydrones/ardupilot/issues/2278

meee1, 2015-05-12:
https://github.com/diydrones/MissionPlanner/commit/86089602341158fee0eb66668660dd1b86752446
added code to read it, still need to add dialogs

meee1, 2015-05-26:
seems like the goal posts are changing here, so im closing this untill there is mroe direction.

closed 2015-05-26 by meee1