ArduPilot/MissionPlanner #868: Show number of accels present on Accel Calibration screen

rmackay9, 2015-05-12:
This came up on the dev call today and the issue is that we've got a flexibility vs safety issue in that we allow the Pixhawk to boot up with less than the normal number of IMUs.

A new pre-arm check has gone into master which ensure that the number of accelerometers being used matches the number of accelerometers that were present during the previous accel calibration.
https://github.com/diydrones/ardupilot/commit/1f14eec4eae065b3a8d985dfc6ac2182b8de2a34

This helps but it doesn't catch the case where the user has a brand-new pixhawk which has less than the expected number of IMUs working.  On the dev call it came up that we should display the number of accels being calibrated on the MP's Accelerometer Calibration page.  There was a suggestion that a picture of the appropriate board type based on the number of accels would be extra nice but perhaps just a number for now.

... but now I wonder if the MP even knows how many accels are present because all the calibration is done inside the flight code..

billbonney, 2015-05-12:
The GCS have no idea how many accels are being calibrated as the procedure is driven by the APM code. In the GCS we only show the level/left/right/nose down/nose up/on back messages

meee1, 2015-05-18:
the only way I see to get around this is to have the code send to the gcs how many it see's