ArduPilot/MissionPlanner #870: Mission Planner and XP

mpjf01, 2015-05-13:
I have been trying to update Mission Planner on my XP computer all day. The update process hangs on the "Check for Updates" dialogue with the message "Getting Updated Parameters" in the body and movement in the horizontal bar. I have tried several times over the last week or so, same result. 

Now I read in RC Groups that "Mission Planner is broken in XP". If true, is this a permanent situation or is it being fixed? I have looked for other reports here and haven't found any so I assume it can't be true. What action do I take to get the update to proceed. 

I have updated my W7 computer successfully but the XP Laptop goes in the field with me so I would like to keep it working.

meee1, 2015-05-13:
can you provide the missionplanner.log file?

mpjf01, 2015-05-13:
How would I best do that? I don't seem to be able to attach it here, even when I zip it.

I have tried to put the files here

https://www.dropbox.com/s/6s02bgggulqfofj/MissionPlanner.log.zip?dl=0

mpjf01, 2015-05-13:
Are these the log files you mean? All the others have dates in the
extensions that are older than today.

mpjf01, 2015-05-15:
I have now tried to update MP on three different XP computers, same problem with each. The upgrade process hangs up part way through, you can see it flashing in the same spot when you do the upgrade in response to an update alert and the command line dialogue stays on the screen. When you upgrade from the MP Help you don't see that dialogue as far as I can tell.

meee1, 2015-05-18:
if you go to this url
what happens?
https://github.com/diydrones/ardupilot/raw/ArduCopter-stable/ArduCopter/Parameters.pde

moeel, 2015-05-18:
I have the same problem here.
when i go to the url with my ie,  it show error "page cannot display". But, if i go the url with my chrome, it routes to another address and open pde text files correctly.
I have tried to reset my ie setting, but no work.

meee1, 2015-05-18:
sounds like you need to update your root certificates on your install.

windows update should be fixing this for you.

moeel, 2015-05-18:
i have all windows update in my laptop, and no work.
I just found that windows xp is not supported anymore for any github link.
https://help.github.com/articles/supported-browsers/ 
i think this is the reason why  i can not open url https://github.com/diydrones/ardupilot/raw/ArduCopter-stable/ArduCopter/Parameters.pde
I check with my chrome that the link is route to https://raw.githubusercontent.com/diydrones/ardupilot/ArduCopter-stable/ArduCopter/Parameters.pde
then, i tried open the route link with my ie8 and it can open the route link and show pde file.
But, I still cannot update my MP.
Maybe if you change the parameter location value directly to the route link, MP can update without problem.
I am not a programmer. Hope you can solve this windows xp not supported issue for all windows XP users.
Thanks.

mpjf01, 2015-05-19:
When I open that URL with FF I get what looks like a program listing for arducopter firmware.

moeel, 2015-05-19:
@mpjf01: did you try open in ie8? When you open https://github.com/diydrones/ardupilot/raw/ArduCopter-stable/ArduCopter/Parameters.pde in FF it route to another address right?

mpjf01, 2015-05-19:
I don't use IE so I can't say what might happen. With FF the URL opens up the program listing.

mpjf01, 2015-05-19:
This is what is shown oin the address bar

https://raw.githubusercontent.com/diydrones/ardupilot/ArduCopter-stable/ArduCopter/Parameters.pde

meee1, 2015-05-20:
if the url doesnt loads in IE, then you need to update your root certificates on your machine.

moeel, 2015-05-20:
Is there any relationship between IE and the update process of MP? I can load the url in my Chrome without problem, but i still can not getting parameters update in my MP.
Another thing is the fact that github stop support for windows xp. Any idea for this?

meee1, 2015-05-20:
please go here https://www.digicert.com/digicert-root-certificates.htm
in IE
and click the "DigiCert High Assurance EV Root CA" test link.
and report back

moeel, 2015-05-22:
I can not open https://www.digicert.com in IE.

meee1, 2015-05-22:
the problem is as ive said all along, your pc is not getting root certificate updates, you need to fix this.
download this https://www.digicert.com/CACerts/DigiCertHighAssuranceEVRootCA.crt
and install it as a trusted root certificate.

moeel, 2015-05-22:
I already tried to install that root certificate before, but no work. I just install it again as you recommend it and no work too.

Anyway, I just updated my MP to 1.3.27, i know that you change the update procedure. Now update process will not stuck on failure of getting update parameters data .pde. It can continue to the next process. Actually what is the effect if the getting update parameter failed?

Did you tried the link on windows xp by yourselves? because I have tried to open the link in other windows xp installation, no work too.

meee1, 2015-05-23:
download https://www.digicert.com/CACerts/DigiCertHighAssuranceEVRootCA.crt using FF or chrome, then open it, select install certificate, pick plase all cert in the following store, select trusted root certificate authority, keep click next.
that should install the cert, and also fix your github display issue in IE

moeel, 2015-05-23:
Yes, I already did what you recommend. But, still can not work.
Please see below my screen shot. It show that the digicert is installed, but still can not access github.

[image]
[image]

Please anyone else with windows XP can confirm this issue.
https://help.github.com/articles/windows-xp-is-not-supported/ this link also confirm that XP is not supported anymore.
https://github.com/blog/1937-improving-github-s-ssl-setup this link stated that we can not open github.com in IE windows XP since January 5, 2015.

meee1, 2015-05-24:
sounds like there is nothing I can do then.
as of version 1.3.27 it will skip the part where it ties to download from github during an update

mpjf01, 2015-05-24:
If you are saying that a future release will support the ability to update MP on XP computers, thank you.

meee1, 2015-05-26:
it should, but not 100%
the only path forward is to move to windows 7

closed 2015-05-26 by meee1

mpjf01, 2015-05-26:
I now get a different error and the update fails. The process is in a loop. 

I can't move to W7, the PCs that I take into the field don't support it.

moeel, 2015-06-08:
I found the solution for XP user. I modified file missionplanner.exe.config and replace the url https://github.com/diydrones/ardupilot/raw/ArduCopter-stable/ArduCopter/Parameters.pde with https://raw.githubusercontent.com/diydrones/ardupilot/ArduCopter-stable/ArduCopter/Parameters.pde . You can use notepad to modify the file. If you use MP for other drone type, you need to replace the other url for plane, etc also. Unfortunately after update, the missionplanner.exe.config will be replaced again to the original one, so i modify it again after update. Hoping this can help for all XP user.

AndyLeeRobinson, 2015-07-22:
I use XP64 - I updated the root certs and edited missionplanner.exe.config, still no joy.
Reinstall required. So much for progress!