ArduPilot/MissionPlanner #877: Add support Arducopter Landing gear

Jman841, 2015-05-21:
Currently Option 29 (landing gear) for RC9 and higher is not displayed as a value in Mission planner, nor is "Landing Gear" an option for CH7 and higher to set this new function using AC 3.3 in mission planner.

meee1, 2015-05-26:
this needs to be raised on the ardupilot issue list, as it needs to be added to the autodoc system for parameters

closed 2015-05-26 by meee1

Jman841, 2015-05-26:
The parameter works if the number is set to "28" in mission planner as described in the wiki for auto landing gear. The issue is mission planner does not state that the value "28" is for auto landing gear nor does it allow for the selection of auto landing gear switch in the basic setup tab for Rc channel out, but, if set manually it does work. Is this still an ardupilot issue or a mission planner issue?