ArduPilot/MissionPlanner #879: Compass learn gets turned off after compass cal

magicrub, 2015-05-22:
COMPASS_LEARN is always set to 0 after a a compass calibration. I have only tested this with the pixhawk GPS+compass module. I can understand it being off during a calibration but the expected behavior would be to restore the value after the calibration.

meee1, 2015-05-26:
MP has always disabled it if the user does a calibration.

magicrub, 2015-05-26:
So it's always been a problem?

meee1, 2015-05-26:
what makes you think its a problem?
im failing to see why its a problem at all.

magicrub, 2015-05-26:
my use case: fresh pixhawk, then upload a param file which sets learn to
true, then calibrate, then go back and set learn to true.

Why not resume that setting?

CraigElder, 2015-05-27:
The normal use case is that there is no calibration so learn is set to ON.
Then somebody comes along and does a calibration and they normally don't want it written over so the planners all set compass learn to OFF.

If we leave the parameter set to ON then on the very next flight their new calibration is overwritten.  People will not be happy.

magicrub, 2015-05-27:
So when learn is enabled a calibration is never needed?

meee1, 2015-05-27:
correct, but is recommended.
in flight the mag is only exposed to -45 - + 45 roll, -20 + 45 pitch? and all of yaw, this is a very one sided sphere. the calibration forces it beyond that narrow scope. and can provide a better instant answer.

magicrub, 2015-05-27:
Sounds like it is best to calibrate then turn learn off. Although if it is
always calibrating for the orientation that it has available then the only
calibration I would need would be for the orientations I use. And learning
as I go always keeps that calibration valid.

So I guess I don't see the benefit of ever calibrating? I'm not opposed to
it, I guess I just don't know the performance difference so to me out
sounds equal but one method requires work and the other doesn't. Free lunch
too good to be true?

meee1, 2015-05-28:
im going to close this, as at this point in time MP is not going to change this behaviour unless some other issue is identified.

closed 2015-05-28 by meee1

magicrub, 2015-05-28:
Is COMPASS_LEARN helpful if you relocate (and fly) your aircraft far away where the magnetic declination is different than where you did a calibration?

meee1, 2015-05-28:
no, because the magnetic field strength may change, but the natural bias the mag has wont change.

rmackay9, 2015-05-29:
...also we have auto declination which is on by default and a separate feature from compass learning