ArduPilot/MissionPlanner #882: PARAM_SET compatibility

olliw42, 2015-05-25:
If I'm not mistaken, when writing parameters to APM, MissionPlaner currently sends out only the PARAM_SET messages, which - for each PARAM_SET - triggers APM to also store the change in its permanent memory (EEPROM).

Could MP's behavior be changed such, that after it has send out all PARAM_SET messages, it in addition sends out a MAV_ACTION_STORAGE_WRITE message (which I couldn't find in the docs, but which seems to be substituted by the MAV_CMD_PREFLIGHT_STORAGE command)?

The reasons are: The above procedure is not Mavlink compatible, the docu clearly says for PARAM_SET: "Set a parameter value TEMPORARILY to RAM. It will be reset to default on system reboot. Send the ACTION MAV_ACTION_STORAGE_WRITE to PERMANENTLY write the RAM contents to EEPROM". Furthermore, there are good practical reasons for doing the store in EEPROM separately after parameters were modified, e.g. when the EEPROM is emulated by flash the maximal number of erase cycles is quite limited, and since always pages need to be erased doing that for each parameter is inefficient.

Since APM seems to ignore the MAV_CMD_PREFLIGHT_STORAGE command, the suggested change would be fully compatible with APM, but would other systems allow to do their job better, and potentially live somewhat longer.

meee1, 2015-05-31:
if you press the shortcut key control-y inside mp it will send a preflight_storage command

olliw42, 2015-06-02:
THANKS a lot for the addition
it is not ideal but provides an option
may I ask, just for my education, why did you stay away from sending a preflight_storage command by default after each parameter write session, I mean, what do I miss?
thx again

meee1, 2015-06-02:
yes. correct, both are implemented.
having the storage write sent after a 10 seconds delay should save a lot of eeprom writes.

closed 2015-06-02 by meee1

ljalves, 2015-07-29:
I don't really like the automatic saving from ram to eeprom/flash.
AlceOSD desn't have eeprom and uses flash for parameter storage. Although it as implemented a simple flash wear algorithm, I would like to only get the settings saved to flash when I'm happy with the osd layout and not every time 10 sec after I change a parameter.

@meee1, It would be nice to at least have an mission planner option to disable the "automatic eeprom/flash save" 10 sec timer.

meee1, 2015-07-29:
the 10 seconds is only after a param change, that is GCS initiates, so don't think its an issue at all.

ljalves, 2015-07-29:
Well... unfortunately for me it is.
I don't want to write to the flash until I have all the OSD layout fully setup - I could be changing a widget position on the screen until I like where it is. Writing flash every time I change its coordinates its not good.

Eeprom can take around 100x more erase/write cycles than flash so for that case its ok.
But since I use program flash for storing the settings I really need to avoid unnecessary erase/writes.

rmackay9, 2015-07-30:
@ljalves, how about putting some code in the OSD so that it doesn't write to it's own flash too often?

ljalves, 2015-07-30:
Sorry, but in my opinion that doesn't make sense. Things are supposed to be simple:
- Write parameter - writes things to RAM
- MAV_CMD_PREFLIGHT_STORAGE writes ram to eeprom/flash

I can't forbid the user to explicitly do execute a MAV_CMD_PREFLIGHT_STORAGE just because one has happened a few seconds ago.

My suggestion was to add an option to enable or disable the "auto-save to flash/eeprom" feature.
Another suggestion I can give is to add an option to control the timer (the 10 second window).

meee1, 2015-07-30:
what is a good value if not 10 seconds?