ArduPilot/MissionPlanner #884: radio calibration screen not using RCMAP first time it's opened

rmackay9, 2015-05-25:
To reproduce the issue switch around two channels (i.e. RCMAP_PITCH =3, RCMAP_THROTTLE = 2).
Start the MP and connect to Pixhawk running AC3.3-rc5 (or whatever).  Go to the radio calibration screen and note the green bars move as if the RCMAP changes had not been done.
Click to Config/Tuning and then back to Radio Calibratoin and note that the green bars have now moved correctly taking the RCMAP into account.
[image]

meee1, 2015-05-28:
i believe this is fixed

closed 2015-05-28 by meee1

andyp1per, 2015-06-03:
Worked for me when I just tried it with .28