ArduPilot/MissionPlanner #886: SITL in Beta Mission planner does not work...

davidbuzz, 2015-05-30:
when trying to run SITL, In the ArduPlane.exe dos box, I get this error on startup:  /sitl/jsbsim/JSBSim.exe: error while loading shared libraries: MSVCR120.dll: can
not open shared object file: No such file or directory

 the full output is: 

Started model jsbsim at -27.5742683410645,152.91259765625,14.1615104675847,0 at
speed 1.0
Starting sketch 'ArduPlane'
Starting SITL input
bind port 5760 for 0
Serial port 0 on TCP port 5760
Waiting for connection ....
bind port 5762 for 2
Serial port 2 on TCP port 5762
bind port 5763 for 3
Serial port 3 on TCP port 5763
/sitl/jsbsim/JSBSim.exe: error while loading shared libraries: MSVCR120.dll: can
not open shared object file: No such file or directory

meee1, 2015-05-30:
please install 
https://www.microsoft.com/en-au/download/details.aspx?id=5555

meee1, 2015-05-30:
sorry this one
https://www.microsoft.com/en-us/download/details.aspx?id=40784

davidbuzz, 2015-06-02:
could MP be setup to automatically install this.... or at least popup a "please install X from http://blah" ?

meee1, 2015-06-02:
already done
https://github.com/diydrones/MissionPlanner/commit/b8eba3936600d5e09b5858daf3aa0fd57c48f53a

closed 2015-06-02 by meee1