ArduPilot/MissionPlanner #890: APM:Plane: "ter_curalt" MP warning request

GrahamDyer, 2015-06-01:
Currently one can set a Mission Planner altitude warning for an aircraft when it goes below a set altitude above the take-off or home altitude. 

Wouldn't it be better to have a warning for altitude above "ter_curalt" (terrain current altitude) which is surely more relevant as that is the height that the aircraft is above the ground at that point? 

Or could an additional warning checkbox be added?

meee1, 2015-06-02:
try the warning manager.
control-f > warning manager.

GrahamDyer, 2015-08-07:
Yes, thanks, works great. Still think that a default warning for "altitude over terrain" is more relevant and important than one for "altitude over home" as the world ain't flat :))

meee1, 2015-08-07:
look at  ter_curalt

GrahamDyer, 2015-08-08:
Yes, that is what I'm asking for as a default warning, not alt over home

GrahamDyer, 2017-05-29:
Request option to use "ter_curalt" (Alt of MAV over current terrain) if available as a _default_ low altitude warning rather than "altitude above home point" as it is currently configured. 

Low "altitude above home point" warning isn't of much use in any kind of undulating or varying terrain.

closed 2018-06-15 by GrahamDyer