ArduPilot/MissionPlanner #892: Moving waypoints all at once

rdesgagn, 2015-06-03:
Sometimes when I plan a mission, I would like to be able to move all the waypoints at once so I don't have to move each of them separately. I don't know if other users already asked for this feature but I think it would increase efficiency at creating mission.

It could be done by right clicking on one of the leg between two waypoints for example and while holding down the mouse button, drag to where you want the new waypoint(s).

meee1, 2015-06-05:
hold down control, and try.

meee1, 2015-06-15:
work?

closed 2015-06-15 by meee1

rdesgagn, 2015-06-15:
Yes, it works. Thanks a lot Michael.

reluctantgithubuser, 2018-09-10:
Hi-
Did the control select feature get removed?
I cannot select multiple waypoints with control drag.
I can draw the blue box around the waypoints but nothing I click allows me to move all of them.
It just moves one waypoint or adds another if I left click inside the blue box.
Win8 pc, MP 1.3.58