ArduPilot/MissionPlanner #895: Add button for MNT_MODE on flight data screen

sgofferj, 2015-06-05:
I have been scratching my head today on a filming trip, wondering why I can't suddenly control the gimbal via the TX. I quickly rewired my IRIS for CH6 passthru in the field. At home I figured it out. When setting up the gimbal, I used the testing method described in the wiki and used the "Point camera here function" which set MNT_MODE to 4 (GPS). The parameter is persistent over a reboot and hence the RC-control didn't work.
Please add a button to conveniently choose MNT_MODE from the flight data screen to allow for easy transitions.

meee1, 2015-06-06:
please look on the action tab on flightdata, you can switch between all the modes.

meee1, 2015-06-06:
the one labelled set mount

closed 2015-06-06 by meee1